WildernessLabs/Meadow_Issues #693: Linker cannot handle System.Text.Json version 8

ctacke, 2024-06-03:
The linker does not like version 8 of `System.Text.Json`

Repro: 

Using blinky, simply add this code
```
public class Thing
{
    public string Foo { get; set; } = "Bar";
}

public override Task Initialize()
{
    Resolver.Log.Info("Initialize...");

    var json = System.Text.Json.JsonSerializer.Serialize(new Thing());
    Resolver.Log.Info(json);
...
}
```
Add a NuGet ref to `System.Text.Json version 8.0.3`
Build and trim the app:
```
> meadow app build -c Debug
> meadow app trim -c Debug
```
You will get hundreds of errors.

```
Trimming failed
ILLink: error IL1001: Failed to process 'resource mscorlib.xml in mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089'. Feature 'remoting' does not specify a 'featurevalue' attribute
resource mscorlib.xml in mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089: warning IL2012: Could not find field 'param_name' on type 'System.ArgumentException'
ILLink: error IL1001: Failed to process 'resource mscorlib.xml in mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089'. Feature 'remoting' does not specify a 'featurevalue' attribute
[blah, blah, blah]
```
Change the reference to `System.Text.Json` version 7.0.0 and it will build and link just fine

adrianstevens, 2024-06-07:
It's probably time to explore a newer version of the linker

adrianstevens, 2024-06-11:
I might need to team up with @alexischr on this one

hartez, 2024-06-19:
Thank you so much for posting this! Between this and #694 I've been pounding my head on the desk trying to figure out why new code won't deploy and why all my dependencies were missing when I deployed from the command line.