WildernessLabs/Meadow_Issues #694: VS Extension suppresses linker errors

ctacke, 2024-06-03:
Following the bug repro for #693 but instead of using the CLI to build, try to deploy the app from VS 2022.  With the CLI v1 extensions installed, you will get an error:
```
17:00:59:505	1>An error occurred during the App deployment process.
17:00:59:505	1>Error:
17:00:59:505	1>An item with the same key has already been added. 
17:00:59:505	1>Stack Trace :
17:00:59:505	1>   at System.ThrowHelper.ThrowArgumentException(ExceptionResource resource)
17:00:59:505	1>   at System.Collections.Generic.Dictionary`2.Insert(TKey key, TValue value, Boolean add)
17:00:59:505	1>   at Meadow.CLI.Core.Devices.MeadowLocalDevice.<>c__DisplayClass83_1.<<DeployApp>g__AddFile|3>d.MoveNext()
```
This error is a follow-on to the fact that the linker has failed but without providing any information to the VS UI.  This makes it seem like the error is somewhere other than in the linker

CartBlanche, 2024-06-03:
@ctacke I'll give this is try tomorrow morning.

adrianstevens, 2024-06-11:
This is fixed in v2.x

CartBlanche, 2024-06-11:
@adrianstevens was the fix extension side or Meadow.CLI side?