WildernessLabs/Meadow_Issues #698: [v2] Meadow.CLI  - meadow app trim not finding firmware packages...

CartBlanche, 2024-06-04:
**Describe the bug**
I have the latest firmware download to `C:\Users\savag\AppData\Local\WildernessLabs\Firmware` and I build and then trim via the command line, on Windows 11. The Trim command is unable to find the firmware, to trimming does not happen.

**To Reproduce**
Run console command:
 `meadow firmware list`
to get:
Firmware location: C:\Users\savag\AppData\Local\WildernessLabs\Firmware
 Meadow F7
    1.9.0.0
    1.10.0.2
    1.11.0.0
  * 1.12.0.0 (default)

Navigate to Meadow project directory and run command:
`meadow app build -c Debug` 
to get:
Building Debug configuration of C:\Users\savag\source\repos\MeadowApplication1\MeadowApplication1...
Build successful

then run command:
`meadow app trim -c Debug`
to get this error message:

No firmware packages found, run 'meadow firmware download' to download the latest firmware

**Expected behaviour**
It should find the firmware package that has previously download and continue trimming the recently built app.

**Screenshots**
N/A

**Developer tools (please complete the following information as best as you can):**
 - Windows:
Device name	Win11Laptop
Processor	AMD Ryzen 7 PRO 6850U with Radeon Graphics        2.70 GHz
Installed RAM	32.0 GB (31.3 GB usable)
Device ID	E38B8074-580D-4282-BFB8-A020D5C952B7
Product ID	00355-60914-51799-AAOEM
System type	64-bit operating system, x64-based processor

Meadow.CLI `v2.0.46`

adrianstevens, 2024-06-04:
Repro'd

CartBlanche, 2024-06-04:
@adrianstevens fix appears to work.

adrianstevens, 2024-06-04:
Fixed in v2.0.47 - closing issue

closed 2024-06-04 by adrianstevens