WildernessLabs/Meadow_Issues #708: CLI 2.0.48 - Writing file to board show incorrect file name

NevynUK, 2024-06-11:
I am trying `meadow app deploy .` running from the binaries directory and the command progress looks to be overwriting the line above when giving progress updates.

Sample output:

```
Sending  'System.Configuration.dll'
Sending  'System.Xml.dll'ion.dll: 100%
Writing System.Xml.dll: 67%
```

CLI appears to be doing the following:

* Issues the Writing message and then updates the progress through to 100%
* Move to the next file
* Carriage return but no line feed
* Picks the next file and then writes the Sending message for the next file
* Newline
* Starts to write the progress for new file
* Rinse and repeat.

adrianstevens, 2024-06-11:
@NevynUK  you're on a Mac I assume?

adrianstevens, 2024-06-11:
And is this new behaviour?

adrianstevens, 2024-06-11:
I'm told it's not a new bug - apparently I just never noticed ;)

NevynUK, 2024-06-11:
> @NevynUK you're on a Mac I assume?

Yes.

adrianstevens, 2024-06-16:
the overwrite behavior is intended ... not cleaning up the remaining characters on the line is not 

fixed in v2.0.49

closed 2024-06-18 by adrianstevens