WildernessLabs/Meadow_Issues #718: Version 1.12.2 Crashes When Loading Settings From Resolver.Services

thegreatco, 2024-06-14:
**Describe the bug**
Calling `Resolver.Services.GetOrCreate<MySettings>()` in the `Initialize` method of `App<F7FeatherV1>` while running 1.12.2 of the Meadow nuget packages causes the Meadow to reset with an (apparently) uncatchable error.

**To Reproduce**
Steps to reproduce the behavior:
1. Add some settings to `app.config.yaml`
2. Create a `MySettings` class to read them, extending the `ConfigurableObject` class
3. In `Initialize` of your `App<F7FeatherV1>`, call `Resolver.Services.GetOrCreate<MySettings>()`
4. Meadow resets with no error/output to console

**Expected behavior**
The settings object is created and registered with the resolver like in 1.12.0.

**Developer tools (please complete the following information as best as you can):**
 - Windows 11
 - Visual Studio Community 2022 Version 17.10.0 Preview 4.0
 - Meadow extension for IDE version: 2.0.0.4

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7FeatherV1
 - Meadow OS version: 1.12.2

**Additional context**
None.

ctacke, 2024-06-14:
I'll investigate, but ConfigurableObject is slow AF on the F7, so I'd recommend swapping to this:
https://blog.wildernesslabs.co/meadow-app-settings/

adrianstevens, 2026-11-07:
Closing: this was reported against the 1.12.2 Meadow packages, which are no longer supported. Please open a new issue if it reproduces on a current release.

closed 2026-11-07 by adrianstevens