WildernessLabs/Meadow_Issues #721: [CLI, meadow app run] Unexpected `%` suffix on deploy lines

patridge, 2024-06-15:
**Describe the bug**
When deploying an app via CLI, I got some weird final output on lines for a given file message line where it would normally show something like `Sending 'App.dll'`.

Sometimes, there was no percent or a colon and space plus `100%`.

```console
Sending  'Meadow.dll'.dll: 100%         
Sending  'System.Core.dll'
```

Sometimes the colon and space disappeared with `100%`.

```console
Skipping 'app.config.yaml'100%
```

Sometimes it was just the percent sign without a space.

```console
Skipping 'System.Memory.dll'%
```

Sometimes it was some weird percent result.

```console
Sending  'System.Xml.dll'0%
...
Sending  'App.pdb'00%
```

This was not consistent between deploys for the same files, skipped or sent.

**To Reproduce**
Steps to reproduce the behavior:
1. On a Meadow app with device plugged in, run the following command from within the project folder: `meadow app run . -c Release`

**Expected behavior**
Lots of file deploy messages about `Skipping` and `Sending` with file names. I suppose I expect `Sending` to have `{filename}: 100%` after them. Skipped files wouldn't need any percent. But I mostly just don't expect the inconsistency.

**Raw build output**

Here's the full output from one such app run.

```console
% meadow app run . -c Release            
Disabling runtime...
Building Release configuration of /.../LedFun for Meadow v1.12.2.0...
Trimming application /.../LedFun/bin/Release/netstandard2.1/App.dll...
Deploying app from /.../LedFun/bin/Release/netstandard2.1...
Found trimmed binaries in post link folder...
Generating list of files to deploy...
Deleting 'Microsoft.Bcl.AsyncInterfaces.dll'
Deleting 'System.Numerics.Vectors.dll'
Deleting 'System.Runtime.CompilerServices.Unsafe.dll'
Deleting 'System.Text.Json.dll'
Deleting 'System.Threading.Tasks.Extensions.dll'
Deleting 'Microsoft.Extensions.Configuration.Json.dll'
Deleting 'mscorlib.pdb'
Deleting 'System.Core.pdb'
Deleting 'System.pdb'
Deleting 'System.Security.dll'
Deleting 'meadow.log'
Deleting 'dns.conf'
Sending  'MQTTnet.dll'
Skipping 'System.Web.dll'         
Sending  'MicroJson.dll'
Sending  'ProjectLab.dll'0%         
Sending  'Meadow.Units.dll'%         
Sending  'Mcp23xxx.dll'l: 100%         
Skipping 'System.IO.Compression.FileSystem.dll'
Sending  'Meadow.F7.dll'
Sending  'netstandard.dll'%         
Sending  'Meadow.Foundation.dll'      
Sending  'System.Configuration.dll'         
Sending  'Meadow.Logging.dll'dll: 100%         
Skipping 'System.Net.Http.dll'0%         
Sending  'Microsoft.Extensions.Configuration.FileExtensions.dll'
Sending  'Microsoft.Extensions.Configuration.dll'tensions.dll: 100%         
Skipping 'System.IO.Compression.dll'ration.dll: 100%         
Sending  'Microsoft.Extensions.FileSystemGlobbing.dll'
Sending  'Microsoft.Extensions.Primitives.dll'g.dll: 100%         
Sending  'NetEscapades.Configuration.Yaml.dll'00%         
Sending  'mscorlib.dll'nfiguration.Yaml.dll: 100%         
Sending  'Microsoft.Extensions.Configuration.Abstractions.dll'
Sending  'System.IO.Hashing.dll'figuration.Abstractions.dll: 100%         
Skipping 'YamlDotNet.dll'.dll: 100%         
Skipping 'System.Buffers.dll'
Sending  'Microsoft.Extensions.FileProviders.Abstractions.dll'
Sending  'Meadow.Contracts.dll'leProviders.Abstractions.dll: 100%         
Skipping 'Mono.Security.dll': 100%         
Sending  'System.dll'
Sending  'Microsoft.Extensions.FileProviders.Physical.dll'
Skipping 'System.Numerics.dll'ileProviders.Physical.dll: 100%         
Sending  'Sc16is7x2.dll'
Sending  'System.Xml.dll'0%         
Sending  'MicroGraphics.dll'         
Sending  'Meadow.dll'.dll: 100%         
Sending  'System.Core.dll'       
Skipping 'System.Memory.dll'%         
Sending  'Apa102.dll'
Sending  'App.dll': 100%         
Sending  'App.pdb'00%         
Skipping 'meadow.config.yaml' 
Sending  'wifi.config.yaml'
Skipping 'app.config.yaml'100%         
Sending  'App.deps.json'
Writing App.deps.json: 100%         
Enabling runtime...
Listening for messages from Meadow...
```

**Developer tools (please complete the following information as best as you can):**
 - OS and version: macOS Sonoma 14.4.1
 - IDE and version: CLI
 - Meadow CLI extension version: 2.0.48

**Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).

On a Project Lab board, but not sure on hardware revision.

```console
Board Information 
    Model: F7Micro
    Hardware version: F7CoreComputeV2
    Device name: MeadowF7

Hardware Information 
    Processor type: STM32F777IIK6
    ID: ...
    Serial number: ...
    Coprocessor type: ESP32
    MAC Address - 
        WiFi: ...

Firmware Versions 
    OS: 1.12.2.0
    Runtime: 1.12.2.0
    Coprocessor: 1.12.2.0
    Protocol: 8
```

**Additional context**
Add any other context about the problem here. If you are working with additional external hardware/sensors, please list them here.

adrianstevens, 2024-06-16:
Ok I can definitely fix this - I'll do it tomorrow

patridge, 2024-06-16:
Possibly related #708

adrianstevens, 2024-06-16:
Duplicate of https://github.com/WildernessLabs/Meadow_Issues/issues/717 

closing

closed 2024-06-16 by adrianstevens