WildernessLabs/Meadow_Issues #723: ESP32 error with latest template(s?): `InvalidConfigurationFile`

patridge, 2024-06-17:
**Describe the bug**
When running an app created from the latest templates (definitely `meadow-startkit`, possibly others), there is an ESP32 error about configuration logged to the output.

> The ESP32 has had an error (InvalidConfigurationFile).

**To Reproduce**
Steps to reproduce the behavior:
1. Install WildernessLabs.Meadow.Template v1.12.2
2. Create a new project (e.g., `dotnet new meadow-startkit`)
3. Run the app on a Meadow device

> Log level: Information
> Device is configured to use WiFi for the network interface
> The ESP32 has had an error (InvalidConfigurationFile).
> All cloud features are disabled.

**Expected behavior**
No errors about default, provided template config.

**Workaround**
It appears this is a result of the meadow.config.yaml file's contents, specifically the `Coprocessor:` and `Network:` lines. If these are empty but the parent object isn't commented out, it will cause that error. Commenting them out makes the error go away.

alexischr, 2024-06-25:
Should be just commenting out `Coprocessor:` and `Network:` in the config file

adrianstevens, 2026-11-07:
Closing: this was reported against the 2024 project templates, which have since been updated. Please open a new issue if a current template still logs InvalidConfigurationFile.

closed 2026-11-07 by adrianstevens