WildernessLabs/Meadow_Issues #725: IWiredNetworkAdapter: NetworkConnected and NetworkDisconnected events not firing

GeorgeStephens, 2024-06-19:
**Describe the bug**
Under `IWiredNetworkAdapter`, should `NetworkConnected `and `NetworkDisconnected `events be raised on plugging in and unplugging the ethernet cable?

**To Reproduce**
Steps to reproduce the behavior:

### MeadowApp.cs
```
public class MeadowApp : App<F7CoreComputeV2>
{
    IWiredNetworkAdapter ethernet;

    public override Task Initialize()
    {
        Resolver.Log.Info("Initialize...");

        //// get the Ethernet adapter
        ethernet = Resolver.Device.NetworkAdapters.Primary<IWiredNetworkAdapter>();

        if (ethernet != null)
        {
            // connected event test.
            ethernet.NetworkConnected += EthernetAdapterNetworkConnected;
        }

        return Task.CompletedTask;
    }

    public override Task Run()
    {
        Resolver.Log.Info("Run...");

        if (ethernet != null)
        {
            ethernet.NetworkConnected += EthernetAdapterNetworkConnected;
            ethernet.NetworkDisconnected += EthernetAdapterNetworkDisconnected;

            Resolver.Log.Info("Ethernet Events registered");

        }

        return Task.CompletedTask;
    }

    void EthernetAdapterNetworkConnected(INetworkAdapter networkAdapter, NetworkConnectionEventArgs e)
    {
        Resolver.Log.Info("Ethernet Connection Up");
    }

    private void EthernetAdapterNetworkDisconnected(INetworkAdapter sender, NetworkDisconnectionEventArgs args)
    {
        Resolver.Log.Info("Ethernet Connection Down");
    }

}
```

### meadow.config.yaml
```
# Acceptable values for true: true, 1, yes
# Acceptable values for false: false, 0, no

# Main Device Config
Device:

    # Name of the device on the network.
    Name: MeadowDevice

    # Uncomment if SD card hardware present on this hardware 
    # (e.g., Core-Compute module with SD add-on)? Optional; default value is `false`.
    SdStorageSupported: true

# Control how the ESP coprocessor will start and operate.
Coprocessor:

    # Should the ESP32 automatically attempt to connect to an access point at startup?
    # If set to true, wifi.yaml credentials must be stored in the device.
    AutomaticallyStartNetwork: true

    # Should the ESP32 automatically reconnect to the configured access point?
    AutomaticallyReconnect: true

    # Maximum number of retry attempts for connections etc. before an error code is returned.
    MaximumRetryCount: 7

# Network configuration.
Network:

    # Network Interface (Default value is WiFi with DHCP)
    Interfaces:
        - Name: Ethernet
          UseDHCP: false
          IPAddress: 192.168.1.60
          NetMask: 255.255.255.0
          Gateway: 192.168.1.254
        - Name: WiFi
          UseDHCP: true
#          IPAddress:
#          NetMask:
#          Gateway:

    # Default Interface
    DefaultInterface: Ethernet

    # Automatically attempt to get the time at startup?
    GetNetworkTimeAtStartup: true

    # Time synchronization period in seconds.
    NtpRefreshPeriod: 600

    # Name of the NTP servers.
    NtpServers:
        - 0.pool.ntp.org
        - 1.pool.ntp.org
        - 2.pool.ntp.org
        - 3.pool.ntp.org

    # IP addresses of the DNS servers.
    DnsServers:
        - 1.1.1.1
        - 8.8.8.8
```

**Expected behavior**
A clear and concise description of what you expected to happen.

I've got my ethernet adapter working and I can get details of the connection such as IP address all fine, but subscribing to the events, they never seem to fire. In the below screenshot you can see that the OS is outputting info (I'm using LogLevel.Trace) but the text is different to what I have in my events.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Developer tools (please complete the following information as best as you can):**
 - OS and version: [e.g., Windows 10 v22H2, macOS Monterey v12.6]
 **Windows 11 23H2**
 - IDE and version: [e.g., Visual Studio Professional 2022 v17.3.5, Visual Studio 2022 for Mac v17.3.5]
 **Visual Studio Professional 2022**
 - Meadow extension for IDE version: [e.g., VS 2022 Tools for Meadow v0.19.9]
**CLI v2.0.51
vs tools 1.9.7
extension 1.12.2** 

**Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: [e.g. F7v2, CCMv2]
 **CCMv2**
 - Meadow OS version: [e.g., v0.6.7.19]
 **1.12.2**

duduita, 2024-06-24:
Would you mind testing with DHCP enabled @GeorgeStephens ? I suspect that it might be a DHCP issue.

```
# Network configuration.
Network:

    # Network Interface (Default value is WiFi with DHCP)
    Interfaces:
        - Name: Ethernet
          UseDHCP: true
          IPAddress: 192.168.1.60
          NetMask: 255.255.255.0
          Gateway: 192.168.1.254
        - Name: WiFi
          UseDHCP: true
#          IPAddress:
#          NetMask:
#          Gateway:
...
```

duduita, 2024-06-24:
Never mind my previous comment, the problem is with your Ethernet `IPAddress`, `Gateway` or `NetMask`, please ensure that you are using the correct values, or, if you don't need a static IP, just enable the `DHCP`, as I mentioned before, then it should work.

alexischr, 2024-06-25:
We will start working backwards from a Meadow.Core definition of events, and normalize behavior of these events as triggered by the OS (e.g. trigger NetworkConnected as soon as the adapter is connected an endpoint, regardless of IP setup, DHCP lease grant and Internet/Intranet access.

GeorgeStephens, 2024-06-29:
So currently the `EthernetAdapterNetworkConnected` will only be triggered if using DHCP? But going forwards the pan is to change this to be fired not based on the DHCP status?

duduita, 2024-06-29:
Let me clarify @GeorgeStephens, at first glance, we thought that it was an issue on the ethernet events, however, I and @halyssonJr confirmed that the events are working perfectly, with and without the DHCP.

I'd say that you are not seeing the events, because you are not connecting to Ethernet, since to connect to Ethernet, you have two options: 

1) Using the DHCP, so you don't need to configure the `IPAddress`, `NetMask` and `Gateway`:

```
    Interfaces:
        - Name: Ethernet
          UseDHCP: true
```

2) Without the DHCP, but in this case, you need to properly set up the `IPAddress`, `NetMask` and `Gateway`:

```
    Interfaces:
        - Name: Ethernet
          UseDHCP: false
          IPAddress: 192.168.1.60
          NetMask: 255.255.255.0
          Gateway: 192.168.1.254
```

However, given that your `IPAddress`, `NetMask` and `Gateway` are the same as in the [Ethernet docs](https://developer.wildernesslabs.co/Meadow/Meadow.OS/Networking/), I'd say that you haven't configured them. 

To sum up, if you don't need a static IP address, you can just enable the DHCP, and let the IP be set automatically. Otherwise, you need to disable the DHCP, but in this case remember to configure properly the `IPAddress`, `NetMask` and `Gateway`, choosing a valid `IPAddress` within your local network, and that's not already in use.

GeorgeStephens, 2024-06-29:
@duduita, thanks for your responce.

I think the issue I have is what @halyssonJr says, _**Otherwise, when we disabled the DHCP the events were registered but nothing happened.**_

I've just created an new app as described in the initial report with the addition of `DisplayNetworkInformation();` taken from one of the samples.

Changing

```
if (ethernet != null)
{
    // connected event test.
    ethernet.NetworkConnected += EthernetAdapterNetworkConnected;
}
```

for 

```
if (ethernet != null)
{
    // connected event test.
    ethernet.NetworkConnected += EthernetAdapterNetworkConnected;

    DisplayNetworkInformation();
}
```

where `DisplayNetworkInformation()` is:

```
public void DisplayNetworkInformation()
{
    NetworkInterface[] adapters = NetworkInterface.GetAllNetworkInterfaces();

    if (adapters.Length == 0)
    {
        Resolver.Log.Warn("No adapters available");
    }
    else
    {
        foreach (NetworkInterface adapter in adapters)
        {
            IPInterfaceProperties properties = adapter.GetIPProperties();
            Resolver.Log.Info("");
            Resolver.Log.Info(adapter.Description);
            Resolver.Log.Info(string.Empty.PadLeft(adapter.Description.Length, '='));
            Resolver.Log.Info($"  Adapter name: {adapter.Name}");
            Resolver.Log.Info($"  Interface type .......................... : {adapter.NetworkInterfaceType}");
            Resolver.Log.Info($"  Physical Address ........................ : {adapter.GetPhysicalAddress()}");
            Resolver.Log.Info($"  Operational status ...................... : {adapter.OperationalStatus}");

            string versions = string.Empty;

            if (adapter.Supports(NetworkInterfaceComponent.IPv4))
            {
                versions = "IPv4";
            }

            if (adapter.Supports(NetworkInterfaceComponent.IPv6))
            {
                if (versions.Length > 0)
                {
                    versions += " ";
                }
                versions += "IPv6";
            }

            Resolver.Log.Info($"  IP version .............................. : {versions}");

            if (adapter.Supports(NetworkInterfaceComponent.IPv4))
            {
                IPv4InterfaceProperties ipv4 = properties.GetIPv4Properties();
                Resolver.Log.Info($"  MTU ..................................... : {ipv4.Mtu}");
            }

            if ((adapter.NetworkInterfaceType == NetworkInterfaceType.Wireless80211) || (adapter.NetworkInterfaceType == NetworkInterfaceType.Ethernet))
            {
                foreach (UnicastIPAddressInformation ip in adapter.GetIPProperties().UnicastAddresses)
                {
                    if (ip.Address.AddressFamily == System.Net.Sockets.AddressFamily.InterNetwork)
                    {
                        Resolver.Log.Info($"  IP address .............................. : {ip.Address}");
                        Resolver.Log.Info($"  Subnet mask ............................. : {ip.IPv4Mask}");
                    }
                }
            }
        }
    }
}
```

This is the output window after I have unplugged and plugged the adapter in a few times. Nothing.

[image]

But I can ping the device and you can see the MAC addresses match.

[image]

I first noticed this on a 'stress test' app we have that is doing HTTP posts as well as polling multiple different devices via Modbus TCP so I know that ethernet is working on the device. The network settings are the same because they work for me. I only have DHCP working in the 100..240 range on my home network as everything we use is static IP'd, you don't want to have devices moving when you are trying to poll them via Modbus, it would be very inconvenient!

duduita, 2024-06-29:
Halysson was facing issues because he used an IP address that wasn't within his local network by accident, after fixing it, he could connect using a static IP address.

Anyway, I am pretty sure that the events are working with DHCP enabled. I'll double-check if the events are working with DHCP disabled as well.

duduita, 2024-06-30:
Would you mind setting your log level to `Trace` @GeorgeStephens? 

You can enable it by adding the following to your `app.config.yaml`:
```
Logging:
 LogLevel:
   Default: Trace
```

Then, please confirm if you are seeing the following when connecting:
`Meadow StdOut: Ethernet InvokeEvent NetworkConnectedEvent returned CompletedOk`

In this case, I believe that now I found out the root cause of the issue. Basically, the network events are triggering in the Meadow.Core, but there's something wrong with the network connection handler.

GeorgeStephens, 2024-07-01:
Great news thank you.

On starting without ethernet cable plugged in:

```
Running App
Run...
Ethernet Events registered
```

Then when I plug the ethernet cable in I get:

```
Ethernet InvokeEvent NetworkConnectedEvent returned CompletedOk
Ethernet connected event triggered!
Interface: eth0: eth0 Ethernet Up
```

Then when I unplug the ethernet cable in I get:
```
Ethernet InvokeEvent NetworkDisconnectedEvent returned CompletedOk
Ethernet disconnected event triggered!
```

Plugging in again gives a repeat of the original plugged in event.

duduita, 2024-07-01:
Thank you for trying it.

Now, I just need to understand why the `NetworkConnected` handler is not working as it should.

GeorgeStephens, 2024-07-01:
Thanks. Might be completely different reason but #736 might be the same type of issue.

duduita, 2024-07-03:
This issue should be fixed in the upcoming release @GeorgeStephens.

To unblock you until then, you can try this workaround:

Replace:

```
ethernet = Resolver.Device.NetworkAdapters.Primary<IWiredNetworkAdapter>();
```

with the following:
```
var networkAdapters = Resolver.Device.NetworkAdapters;
ethernet = networkAdapters[networkAdapters.Count - 1];
```