WildernessLabs/Meadow_Issues #728: Cultivar stops responding to Meadow.Cloud Commands after some time

bryancostanich, 2024-06-24:
**Describe the bug**
While Cultivar now runs reliably for weeks at a time, it seems to stop responding to Meadow.Cloud commands after some time.

**To Reproduce**
Steps to reproduce the behavior:
1. Run [Cultivar](https://github.com/WildernessLabs/Cultivar).
2. Let it run for a few days.
3. Go to MeadowCloud.co and send the following command with name `Fan`:
```JSON
{
   "IsOn" : true
}
```

**Expected behavior**
The `Vents` should light up on the display.

**Developer tools (please complete the following information as best as you can):**
n/a

**Meadow (please complete the following information as best as you can):**
- Project Lab
```
Board Information 
    Model: F7Micro
    Hardware version: F7CoreComputeV2
    Device name: Cultivar

Hardware Information 
    Processor type: STM32F777IIK6
    ID: 36-00-2E-00-07-51-38-34-35-31-38-33
    Serial number: 336631753438
    Coprocessor type: ESP32
    MAC Address - 
	WiFi: 4C:75:25:D5:28:78

Firmware Versions 
    OS: 1.12.1.1
    Runtime: 1.12.1.1
    Coprocessor: 1.12.1.0
    Protocol: 8
```

bryancostanich, 2024-08-27:
May no longer be an issue.

duduita, 2024-09-22:
After Cultivar had been running for two weeks, I sent a cloud command, and I received the following in the CLI:

```
Meadow StdOut: [cloud] MQTT message received at topic: 36519cc9a60c4d288e49dfef00464e76/commands/2E-00-25-00-12-50-33-53-32-34-31-20
Meadow StdOut: [cloud] Meadow command received
Meadow StdOut: [cloud] Processing Meadow command of type 'Fan'...
Meadow StdOut: [application] Received fan control: True
```

I haven't checked if the Vents had lit up on the display, but considering that I received the MQTT message, this issue looks fixed to me.

closed 2024-09-22 by bryancostanich