WildernessLabs/Meadow_Issues #729: PWM on `D05` on F7Featherv2 does nothing

ctacke, 2024-06-24:
Create an app that does this:

```
var pwm = Device.Pins.D06.CreatePwmPort(50.Hertz());
pwm.Start();
```

Scope says the line is flatlined - no pulses at all.

An `ioctl` is properly being sent from Core with the following:
Channel: 1
Timer: 3
Duty: 32767
Frequency: 50

Peter-Moody, 2024-07-15:
For F7FeatherV2 the F7's Timer 3, Channel 1 is shared by `D05` (PB4) and `D09` (PC6). PWM does work on D09. Looks like the decision was made to support `D09`and not `D05` for PWM.
Behavior
Attempting to use `D05` just silently doesn't work. However, attempting to use `D06` throws an exception. `D05` should do the same.

patridge, 2024-07-25:
This seems to have highlighted some inconsistencies on Docs.

We have this [generic-named pinout diagram](https://github.com/WildernessLabs/Documentation/blob/develop/docfx/Common_Assets/Meadow_F7_Micro_Pinout.svg) (technically v1, I think) that says D05 is PWM-capable.

Then, we have two v2-specific pinout diagrams that differ based on the pinouts in that area of the board.

* [docs/Common_Files/Meadow_F7v2_Micro_Pinout.svg](https://github.com/WildernessLabs/Documentation/blob/develop/docs/Common_Files/Meadow_F7v2_Micro_Pinout.svg) shows _NO_ PWM on D05 or D06.
* [static/Common_Files/Meadow_F7v2_Micro_Pinout.svg](https://github.com/WildernessLabs/Documentation/blob/develop/static/Common_Files/Meadow_F7v2_Micro_Pinout.svg) shows _YES_ to PWM on D05 and D06.

We also have [mention of what I think is this same issue in the Beta6 release notes](https://github.com/WildernessLabs/Documentation/blob/develop/docs/Meadow/Release_Notes/Beta6/index.md?plain=1#L179).

@Peter-Moody Trying to put together a list of docs changes needed, if you can verify. Then, based on that, we've got a lengthy to-do list on our end.

Questions:

* Is D05 NOT capable of PWM on Feather v2 but IS capable on Feather v1?
* Is D06 capable of PWM on Feather v1 or v2?
* Is D05/D06 valid for PWM on other Meadow boards? (Trying to decide what samples to change or note v2 restrictions, such as [M.F.Leds.PwmLed](https://github.com/WildernessLabs/Documentation/blob/develop/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.PwmLed.md?plain=1#L38).)

Action items for docs:

* [ ] Unify on a single, correct, Feather v2 pinout SVG, delete one and [update all references to the remaining](https://github.com/search?q=repo%3AWildernessLabs%2FDocumentation+Meadow_F7v2_Micro_Pinout.svg&type=code), correct one. (This can probably be done by hand-editing the SVG.)
* [ ] [Clarify "v1" subject in the Feather v1 diagram](https://github.com/WildernessLabs/Documentation/blob/develop/docfx/Common_Assets/Meadow_F7_Micro_Pinout.svg), possibly just editing the title text in the raw SVG.
* [ ] Review and update [various usages of the original SVG](https://github.com/search?q=repo%3AWildernessLabs%2FDocumentation%20Meadow_F7_Micro_Pinout.svg&type=code) based on the surrounding text context for v1 or v2, defaulting to v2 whenever it's not specified.
* [ ] Update any Meadow.Foundation.Leds.*.md samples with a note about pin restrictions on Feather v2.
    * [ ] https://github.com/WildernessLabs/Documentation/blob/develop/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.PwmLed.md?plain=1#L37-L38
    * [ ] https://github.com/WildernessLabs/Documentation/blob/develop/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.LedBarGraph.md?plain=1#L37-L38
    * [ ] https://github.com/WildernessLabs/Documentation/blob/develop/docs/api/Meadow.Foundation/Meadow.Foundation.Leds.PwmLedBarGraph.md?plain=1#L37-L38
* [ ] Update [chart in the Meadow_Basics/IO page](https://github.com/WildernessLabs/Documentation/blob/develop/docs/Meadow/Meadow_Basics/IO/index.md?plain=1#L188-L189) to remove PWM channel for D05 and D06.
* [ ] Update [PWM sample here](https://github.com/WildernessLabs/Documentation/blob/develop/docs/Meadow/Release_Notes/Beta3/index.md?plain=1#L697) to a pin that is more ubiquitous than D05.
* [ ] [extra credit] Update this [Fritzing diagram](https://github.com/WildernessLabs/Documentation/blob/develop/docs/API_Assets/Meadow.Foundation.Leds.PwmLedBarGraph/PwmLedBarGraph_Fritzing.png) to avoid D05/D06 use for PWM bar graph

patridge, 2024-07-26:
Possible impact of locations from changes coming in #776.

Peter-Moody, 2024-07-26:
What I noticed was that the F7FeatherV2 image of the board showed both D05 and D06 supported PWM. Neither one do. Our website text was correct. I think that @jorgedevs has fixed the image, but I could be wrong.

Peter-Moody, 2024-07-26:
I have no reason to believe that there is anything wrong with the F7FeatherV1 documentation.

jorgedevs, 2024-07-27:
@patridge D05 and D06 on the F7v1 does work. Is just that v2 they added more features, so drawback was dropping PWM capability to D05 and D06.

patridge, 2024-08-05:
We should have everything addressed now, removing that as a PWM option everywhere we can.

closed 2024-08-05 by patridge