WildernessLabs/Meadow_Issues #731: TimeSpan on F7/Mono only has millisecond resolution

ctacke, 2024-06-24:
TimeSpan on .NET 8 has (at least) microsecond resolution.  On the F7/Mono, everything is rounded to millisecond precision, which kills the ability to use it for PWMs that have microsecond resolution.

This shows the problem very obviously:

```
int micros = 0;

while (true)
{
    var m = micros / 1000d;
    var ts = TimeSpan.FromMilliseconds(m);
    Resolver.Log.Info($"{m}: {ts} = {ts.TotalMilliseconds}");
    await Task.Delay(100);
    micros += 10;
}
```

alexischr, 2024-07-03:
Fix released with 1.12.6.4

alexischr, 2024-07-03:
This should fix button events being dropped by the debouncer logic in Meadow.Core's `DigitalInterruptPort.cs` , when said logic got faster

alexischr, 2024-09-23:
Fixed with https://github.com/WildernessLabs/Meadow/pull/630

alexischr, 2025-09-26:
Fix re-relased with 2.3.0

closed 2025-09-26 by alexischr