WildernessLabs/Meadow_Issues #732: Meadow.Desktop - Template cant run on Linux

jorgedevs, 2024-06-24:
When creating a Meadow.Desktop project Templates on Linux, `Device.Display` is `null`.

Locally, I removed an `ifdef NET7` so `Device.Display` is initialized with .NET 8. Trying to run the app again, im getting more errors about skiasharp not loading correctly:

```console
libdecor-gtk-WARNING: Failed to initialize GTK
Failed to load plugin 'libdecor-gtk.so': failed to init
No plugins found, falling back on no decorations
App Error in App Run: Unable to load shared library 'libSkiaSharp' or one of its dependencies.
```

alexischr, 2024-06-25:
Gonna try this on Mac first, as I don't have a linux box right now.

gerald-romero, 2025-04-22:
10 months and no traction? Same behavior on fresh Ubuntu 22.04 and 24.04 installs. Works fine in Windows.