WildernessLabs/Meadow_Issues #734: CCM Ethernet Port Status

GeorgeStephens, 2024-06-26:
**Is your feature request related to a problem? Please describe.**
For the ethernet port can the status of the two ports be broken out?

**Describe the solution you'd like**

something like

```
//// get the Ethernet adapter
ethernet = Resolver.Device.NetworkAdapters.Primary<IWiredNetworkAdapter>();
var eth0 = ethernet as LAN3954; // assuming you will support different adapters in the future?
Console.WriteLine($"Port 0 Link is {eth0.Port[0].Link}; speed is {eth0.Port[0].speed}");

eth0.LinkChanged += (s, e) =>
{
    Console.WriteLine($"Port {e.Port} Link is {e.Link}; speed is {e.speed}");
};
```

**Additional context**
We can already see the status of ports via LED's on the hardware but having this logged / available within the app would be great.

Peter-Moody, 2024-07-01:
The status is handled by Nuttx. Our code combines these 2 status values into a single value when notified of a status change (see `nuttx/configs/stm32f777zit6-meadow/src/hcom_nx/hcom_nx_startup_mgr.c@135`).

GeorgeStephens, 2024-07-01:
Could port status be queried separately via MDIO?

Peter-Moody, 2024-07-18:
This feature request could be done by adding to the ISR in `/Meadow/nuttx/configs/stm32f777zit6-meadow/src/ethernet/meadow_ethnet_monitor.c` 2 bools, one for each Ethernet port state. This ISR currently queries the dual port LAN chip for the status of each LAN port. After adding the necessary infrastructure, these values would be polled by `Meadow.Core` whenever it received an Ethernet state change event. `Meadow.Core` would then expose these as necessary to the `MeadowApp`.