WildernessLabs/Meadow_Issues #735: Operation not supported or unimplemented on this platform

zwiglm, 2024-06-26:
meadow firmware write
with

CLI:  2.0.51
DFU 0.11

currently 1.12.0 on the F7FeatherV2

PS C:\Users\thinkingUser> dotnet --list-sdks
5.0.416 [C:\Program Files\dotnet\sdk]
6.0.423 [C:\Program Files\dotnet\sdk]
8.0.106 [C:\Program Files\dotnet\sdk]

adrianstevens, 2024-07-09:
Hi @zwiglm - this is an error message from dfu-util unfortunately. This may be an issue with the USB cable or the specific USB port. 

In my experience, devices will deploy successfully but may not flash without being connected directly to your development machine (e.g. no USB hub) or require a different cable.

And we'll look into providing a more meaningful error message in this situation.

zwiglm, 2024-07-09:
Hey, thanks for getting back to me regarding this issue.
Unfortunately, I am on a business trip and can only try out a different cable mid-next week. Unfortunately, I had no other cable on hand at the time of trying, but that was also on my list.
I'll let you know next week, and then I'll close the issue.

zwiglm, 2024-07-17:
Hi. Reporting back.

My go-to cable always used to work on the same explicit USB port. 
The same exact setup does not work anymore.
IMO can therefore exclude cable or port issue

Kind regards

zwiglm, 2024-07-17:
How does one write with dfu-util?
So that I get better debug/error information

adrianstevens, 2024-07-24:
Are you on Windows? If so, are you seeing the device as STM32 BOOTLOADER in the Device Manager? 

https://developer.wildernesslabs.co/Meadow/Getting_Started/Updating_Meadow/

zwiglm, 2024-07-25:
"STM Device in DFU Mode"

zwiglm, 2024-07-25:
So, I reinstalled the driver, and that fixed the problem.
Nevertheless - how it got deleted/overwritten is unknown to me in the first place.

closed 2024-07-25 by zwiglm