WildernessLabs/Meadow_Issues #738: Network adapters have incorrect state at startup

ctacke, 2024-07-03:
When the `F7EthernetNetworkAdapter` class is created the cable state is unknown, so we initialize `IsConnected` to `false`.  Once Core is up, events properly track connected/disconnected state, but if the device is started with the cable connected, no event happens to prime the property in the true state and it will incorrectly return `false`.

We need the ESP to either send an event when Core is up or provide a way for Core to query the state (or trigger the ESP to check state and raise the event)

This may also be an issue for Cell and WiFi if the ESP starts the adapter and it connects before Core is up.

NevynUK, 2024-07-04:
I have checked DHCP and static IP and started the application both with and without the cable attached and I can confirm that the messages do arrive in `EventHandlerServiceThread`.

I have also checked WiFi and the events do arrive but the `Connect` call is not given enough time for the operation to be executed.  If you add the following to the application

```csharp
Device.PlatformOS.BeforeReset += () =>
{
    Console.WriteLine("BeforeReset event.");
    while (true)
    {
        Thread.Sleep(1000);
    }
};
```

then you can see the `Connected` event arriving in an application.

NevynUK, 2024-08-06:
> Mark, I think you pasted the wrong code there

Incorrect - my comment is as follows:

**the events do arrive but the Connect call is not given enough time for the operation to be executed. If you add the following to the application**

followed by:

**then you can see the Connected event arriving in an application.**

I was illustrating how the events are being raised but the application is not running long enough to receive the event.

> I think we should implement a direct pinvoke counterpart to `IsConnected` in the OS that the property can call and get the current status.

I also think this is incorrect.  We already have an `IsConnected` property and this should be using the .NET APIs to check for an IP address.

halyssonJr, 2024-08-18:
@alexischr, I will investigate and try to solve this issue.

alexischr, 2025-01-10:
Released with 2.0

closed 2025-01-10 by alexischr