WildernessLabs/Meadow_Issues #739: CLI Continuously repeats the last message from the application

NevynUK, 2024-07-04:
I have an application deployed to PLv3 and CLI is listening to the board.

The OS is stopped as I am working with the STM32 debugger so there is no output from the application.

CLI seems to continuously repeat the last message received from the appliction:

```
WiFi adapter state changed to: Connecting
Sending command to connect
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
Wifi InvokeEvent NetworkConnecting returned CompletedOk
```

NevynUK, 2024-07-04:
This appears to be true event when not debugging.  Writing a new RT to the board shows the following;

```
Download of 'Meadow.OS.Runtime.bin' success (checksums calculated:0xC8EC240A, expected:0xC8EC240A)
Erasing mono flash memory
Erasing mono flash memory
Erasing mono flash memory
Erasing mono flash memory
Erasing mono flash memory
Erasing mono flash memory
Erasing mono flash memory
Erasing mono flash memory
Erasing mono flash memory
Mono memory erase success
```

I would expect only one line for the message `Erasing mono flash memory`

adrianstevens, 2024-07-04:
Interesting ... I haven't seen this behavior. I'll try to reproduce.

adrianstevens, 2024-07-04:
Repro'd - we'll fix

adrianstevens, 2024-07-09:
fixed in the latest release - closing issue

closed 2024-07-09 by adrianstevens