WildernessLabs/Meadow_Issues #740: Meadow.OS - TimeSpan issue

jorgedevs, 2024-07-05:
I have a my [WiFiWeather](https://github.com/WildernessLabs/Meadow.Samples/tree/jorge-updates/Source/MultiPlatform/WifiWeather) Sample that has an infinite While loop, where in each cycle, i wait for one minute to either do an API call (to get Weather data) or just update the date time on my display.

[image]

I added a Console Output on each cycle, and I found out that the **while loop iterates every 5 seconds approx and not every minute** like I have coded in my app.

adrianstevens, 2024-07-05:
@jorgedevs  can you paste the loop code here?

adrianstevens, 2024-07-05:
I assume it's this? `await Task.Delay(TimeSpan.FromMinutes(1));` 

I'm curious to know if `Thread.Sleep` works

adrianstevens, 2024-07-05:
And try it without a TimeSpan ... just pass an int for milliseconds

jorgedevs, 2024-07-05:
@adrianstevens excellent questions, thank you. I answered those questions updating the issue to give more context.

ctacke, 2024-07-09:
Since the test app above is incomplete, I created a new one (also attached).  This code, running on 1.12.2, works fine.

```
public class MeadowApp : App<F7FeatherV2>
{
    public override Task Run()
    {
        Resolver.Log.Info($"Device running {Device.Information.OSVersion}");

        Task.Run(async () =>
        {
            while (true)
            {
                Resolver.Log.Info("TICK [Task.Delay FromMinutes]");
                await Task.Delay(TimeSpan.FromMinutes(1));
            }

        });

        Task.Run(async () =>
        {
            while (true)
            {
                Resolver.Log.Info("TICK [Task.Delay FromSeconds]");
                await Task.Delay(TimeSpan.FromSeconds(60));
            }

        });

        new Thread(() =>
        {
            while (true)
            {
                Resolver.Log.Info("TICK [Thread.Sleep FromSeconds]");
                Thread.Sleep(TimeSpan.FromSeconds(60));
            }
        }).Start();

        new Thread(() =>
        {
            while (true)
            {
                Resolver.Log.Info("TICK [Thread.Sleep FromMinutes]");
                Thread.Sleep(TimeSpan.FromMinutes(1));
            }

        }).Start();

        return base.Run();
    }
}
```

closed 2024-07-09 by ctacke