WildernessLabs/Meadow_Issues #742: [CLI 2.0.53] `meadow listen` does not reconnect to Meadow F7 Feather on reset

NevynUK, 2024-07-06:
Apparently CLI is supposed to reconnect to the board after a failure, I could not get this to work.

1. Deploy a modified blinky to the board.  The modification I made was to output a message using `Resolver.Log.Info` on every LED change.
2. Enable the application.
3. Execute `meadow listen` and wait for the board to start outputting message from the colour cycle code
4. Reset the board

### Expected behaviour

CLI should eventually reconnect to the board and start showing messages from the application

### Actual behaviour

CLI did not show any output 3 minutes after the board had reset.

adrianstevens, 2024-07-09:
@NevynUK  do you know when this last worked?

NevynUK, 2024-07-09:
I have never had it work but was informed on a call last week that it should and to file an issue if it did not work.

adrianstevens, 2025-04-13:
There's no logic in the CLI today for listen to reconnect after a device reset. Changing this to a feature request. 

I'd love to know if this is commonly needed vs a nice-to-have.