WildernessLabs/Meadow_Issues #744: St7789 Appears to hang

GeorgeStephens, 2024-07-09:
**Describe the bug**
St7789 appears to hang the thread it is running on intermittently. This can be during first write or during the 100th write. 

**To Reproduce**
This always fails before it reaches 350 updates, sometimes it doesnt make the first update. Following application runs without changes on `Core-Compute Breakout v2b`

```
using Meadow;
using Meadow.Devices;
using Meadow.Hardware;
using Meadow.Peripherals.Displays;
using Meadow.Units;
using System.Threading;
using System.Threading.Tasks;
using Meadow.Foundation.Displays;
using System;
using Meadow.Foundation.Graphics;
using Meadow.Foundation.Graphics.MicroLayout;

namespace Meadow.MicroLayoutTest
{
    public class MeadowApp : App<F7CoreComputeV2>
    {
        public IPixelDisplay display;
        public ISpiBus spi5;
        private Label connections;
        private Label memory;
        DisplayScreen displayScreen;

        public override Task Initialize()
        {
            Resolver.Log.Info("Initialize...");

            Resolver.Log.Trace("Instantiating SPI5 Bus");
            spi5 = Resolver.Device.CreateSpiBus(
                Device.Pins.SPI5_SCK,
                Device.Pins.SPI5_COPI,
                Device.Pins.SPI5_CIPO,
                new Frequency(48000, Frequency.UnitType.Kilohertz)
            );
            Resolver.Log.Trace("SPI5 Bus up");

            Thread.Sleep(50);

            Resolver.Log.Trace("Instantiating Display");
            var chipSelectPort = Device.Pins.PD5.CreateDigitalOutputPort(); // project CCM Breakout v2b
            var dcPort = Device.Pins.PA10.CreateDigitalOutputPort();
            var resetPort = Device.Pins.PC8.CreateDigitalOutputPort();

            Thread.Sleep(50);

            display = new St7789(
            spiBus: spi5!,
                chipSelectPort: chipSelectPort,
                dataCommandPort: dcPort,
                resetPort: resetPort,
                width: 240, height: 240,
                colorMode: ColorMode.Format16bppRgb565)
            {
                SpiBusMode = SpiClockConfiguration.Mode.Mode3,
                SpiBusSpeed = new Frequency(48000, Frequency.UnitType.Kilohertz)
            };
            ((St7789)display).SetRotation(RotationType.Normal);

            Thread.Sleep(50);
            Resolver.Log.Info("Starting Display...");

            displayScreen = new DisplayScreen(display)
            {
                BackgroundColor = Color.AliceBlue
            };

            displayScreen.BeginUpdate();

            displayScreen.Controls.Add(
            new Box(0, 0, displayScreen.Width / 2, displayScreen.Height / 2)
            {
                    ForeColor = Color.Red
            },
            new Label(0, 0, displayScreen.Width / 2, displayScreen.Height / 2)
            {
                    Text = "Hello World!",
                    TextColor = Color.White,
                    VerticalAlignment = VerticalAlignment.Center,
                    HorizontalAlignment = HorizontalAlignment.Center
                }
            );

            Color buttonColor = Color.FromHex("EF7D3B");
            Font12x20 font12X20 = new Font12x20();

            connections = new Label(1, 1, displayScreen.Width - 2, 10)
            {
                Text = "0/1",
                TextColor = buttonColor,
                Font = font12X20,
                HorizontalAlignment = HorizontalAlignment.Right
            };
            memory = new Label(1, 16, displayScreen.Width - 2, 10)
            {
                Text = "0/1",
                TextColor = buttonColor,
                Font = font12X20,
                HorizontalAlignment = HorizontalAlignment.Right
            };
            displayScreen.Controls.Add(new[] { connections, memory });

            displayScreen.EndUpdate();

            Resolver.Log.Info("Display up...");

            return Task.CompletedTask;
        }

        public override Task Run()
        {
            Resolver.Log.Info("Run...");

            Resolver.Log.Info("Hello, Meadow Core-Compute!");

            DateTime lastUpdated = DateTime.Now;
            int counter = 0;

            while (true)
            {
                Thread.Sleep(500);

                displayScreen.BeginUpdate();

                connections.Text = $"{counter}";
                if (counter > 500)
                {
                    counter = 0;
                    displayScreen.Invalidate();
                }
                else
                    counter++;

                //long memUsed = GC.GetTotalMemory(false);
                //memory.Text = $"Used:{memUsed:N0}";
                
                if (lastUpdated.Minute != DateTime.Now.Minute)
                {
                    lastUpdated = DateTime.Now;
                    //displayController!.UpdateDateTime(lastUpdated);
                }
                displayScreen.EndUpdate();
            }

            //return Task.CompletedTask;
        }
    }
}
```

**Expected behavior**
The display not to hang.

I'm pretty sure this is an issue with the St7789 in some way. I don't think its hardware related, I have the same issue on Core-Compute Breakout v2b, project lab v1 and custom hardware. The whole thread that is doing the display update just hangs, everything else is okay (all other threads are running, Ethernet okay etc.). Location in the application is completely random, can be first time I try to draw, it can be the 100th time.

**Screenshots**
[image]

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11 v23H2
 - IDE and version: Visual Studio Community 2022 v17.10.2
 - Meadow extension for IDE version: 1.9.8

**Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: `Core-Compute Breakout v2b`
 - Meadow OS version: 
    OS: 1.12.2.0
    Runtime: 1.12.2.0
    Coprocessor:
    Protocol: 8

adrianstevens, 2024-07-09:
@GeorgeStephens try running it at 24Mhz. In our testing, most of these displays aren't stable at 48mhz (and it's possible the traces on the PCB aren't suitable for 48mhz).

GeorgeStephens, 2024-07-09:
@adrianstevens, 24Mhz seems to have helped. Not sure why the CS would be dragged low due to this but the test I have just started it has outlasted anything prior. I think the 48Mhz came from a sample so might just need to be checked what else is set at that.

GeorgeStephens, 2024-07-09:
Spoke too soon, first test got far further, still froze. Second test got to 72.

adrianstevens, 2024-07-24:
Yeah this definitely should work. I need to find my CCM breakout board. It's possible there's an issue with the PCB design. Does 12Mhz improve stability?