WildernessLabs/Meadow_Issues #746: Meadow.CLI -- Command to delete all files times out

jorgedevs, 2024-07-11:
I had trouble with a project lab that went through many changes (OS versions and samples running on `develop` and `NuGet`), so i was recommended to do a `meadow file delete all` to have a clean slate to deploy an app.

When running such command, after a while of deleting files (shown in an output), i would eventually get this message:

```console
The operation has timed out.
```

I had to run the command 3 times to fully delete all the files on my Project Lab.

adrianstevens, 2024-07-24:
So the CLI is running these as individual delete commands. I wonder if we should add a delay between deletes ...

adrianstevens, 2024-07-24:
Added a 100ms delay between files in v2.0.54

adrianstevens, 2024-10-16:
Fixed - closing issue

closed 2024-10-16 by adrianstevens