WildernessLabs/Meadow_Issues #750: 'meadow file read mscorlib.dll' eventually leads to "The operation has timed out."

alexischr, 2024-07-22:
**Describe the bug**

```
$ meadow file read mscorlib.dll
Getting file 'mscorlib.dll' from device...
Received 8180 bytes
Received 16360 bytes
Received 24540 bytes
Received 32720 bytes
Received 40900 bytes
Received 49080 bytes
Received 57260 bytes
Received 65440 bytes
Received 73620 bytes
...
Received 1761168 bytes
Received 1768001 bytes
Received 1774823 bytes
Index was outside the bounds of the array.
Index was outside the bounds of the array.
Received 1781952 bytes
Received 1788835 bytes
Received 1795691 bytes
Received 1802570 bytes
Received 1809401 bytes
Index was outside the bounds of the array.
Index was outside the bounds of the array.
Received 1816193 bytes
Received 1822386 bytes
...
The operation has timed out.
```

```
$ meadow --version
v2.0.53
```

adrianstevens, 2024-07-23:
First attempt at testing and it worked here reading mscorlib.dll 

```
Received 2368797 bytes
Received 2376977 bytes
Received 2385157 bytes
Received 2393337 bytes
Received 2401517 bytes
Received 2409697 bytes
Received 2417877 bytes
Received 2424585 bytes
File written to 'C:\Users\adria\mscorlib.dll'
Success
```

Windows 11
Project Lab v3 running v1.12.6.4

adrianstevens, 2024-07-23:
@alexischr  how consistent is this for you?

adrianstevens, 2024-07-26:
Apparently seen on macOS

adrianstevens, 2025-01-12:
I think we've fixed this now - @alexischr  this was related to the debugging changes iirc

Closing issue - re-open if you're still seeing it

closed 2025-01-12 by adrianstevens