WildernessLabs/Meadow_Issues #754: `meadow app deploy` does not delete files in subdirectories

NevynUK, 2024-08-01:
CLI 2.0.54

I have two crash report files in the `crash` directory.  `meadow app deploy` reports the following when deploying an application:

```
Did not find trimmed binaries folder...
Generating list of files to deploy...
Deleting 'mono_error.txt'
Deleting 'oscrash_report.txt'
```

Running `meadow file list crash` reports the following:

```
Getting file list from '/meadow0/crash/'...
mono_error.txt
oscrash_report.txt
	2 file(s)
```

I have checked the contents of the files and they are old crash reports.

adrianstevens, 2024-08-06:
the CLI meadow deploy command does delete files in sub folders ... I wonder if that folder is protected

adrianstevens, 2024-08-06:
found the bug - fixed for v2.0.55

closed 2024-08-06 by adrianstevens