WildernessLabs/Meadow_Issues #759: OV5642 ArduCAM driver request

upswing1, 2024-08-09:
Earlier on there was a driver available. It is no longer available. It would be great to finish it and include it in foundation

ctacke, 2024-08-09:
It was removed from the develop branch because it didn't work and the tooling we use for release just builds everything.  There's still a branch for it.

On the positive side, recent fixes we made in the SPI driver seem to have fixed whatever was wrong with it.  @adrianstevens did some initial testing last week or earlier this week and was getting valid data from it, so it's now probably just a matter of clean-up and documentation.

adrianstevens, 2024-08-09:
Hopefully we'll have this working in the next week or two

upswing1, 2024-08-09:
Great news

adrianstevens, 2025-03-14:
We've shipped Arducam Mini 2MP support which uses the OV2640.

It shouldn't be a big lift now to add OV5642 support.