WildernessLabs/Meadow_Issues #762: CAN Controller Query with timeout

upswing1, 2024-08-10:
The Can Controller could send a query with an expected response time. This will let know if the data was received for the given query

ctacke, 2024-08-14:
@upswing1  I need more context here.  Are you asking for the ability to send an RTR and wait for a response for it all in one awaitable call?

upswing1, 2024-08-31:
I am asking for the ability for the CAN controller to send queries with different timeouts and expect a response within a certain time frame. If the response is not received within the allocated time, it should trigger an event.  This would allow for better error handling etc. The mechanism could also include retry logic when it fails to be received within the timeout.