WildernessLabs/Meadow_Issues #764: Cell network daemon always runs for a board configured for a cell application

NevynUK, 2024-08-19:
The cell networking daemon is always running on a board with a a cell application deployed even if the application is disabled.

Reproduction steps:

* Deploy a cell application
* Test the application (enabled)
* Disable the application

Monitor the application using `meadow listen` and periodic messages are generated saying `Cell connect script failed, retrying`.

Note that this also happens during a firmware upgrade.

duduita, 2024-08-19:
I think I can just add a conditional [here](https://github.com/WildernessLabs/Meadow/blob/d992edd00291f9342ead4c38c8302f4bc2aecea7/apps/examples/hcom/hcom_startup_manager.c#L451), to only starts the cell PPPD if Mono starts (perhaps using `hcom_mono_ctrl_should_mono_run`), what do you think about it?

NevynUK, 2024-08-19:
That makes sense for this case.

closed 2024-10-29 by alexischr