WildernessLabs/Meadow_Issues #765: Garbage events in the ESP networks events queue

duduita, 2024-08-19:
**Describe the bug**
Garbage events are being delivered in the ESP network events queue

```
Meadow StdOut: [esp] Event received.
Meadow StdOut: [esp] Processing event.
Meadow StdOut: [esp] Simple event, interface 1, event code: 38, status code 0x00000000
Meadow StdOut: [esp] Event data collected, raising event.
Meadow StdOut: [esp] Waiting for event.
Meadow StdOut: [esp] Event received.
Meadow StdOut: [esp] Processing event.
Meadow StdOut: [esp] Simple event, interface 48, event code: 2099147, status code 0x00000000
Meadow StdOut: [esp] Event data collected, raising event.
Meadow StdOut: [esp] Waiting for event.
Meadow StdOut: [esp] Event received.
Meadow StdOut: [esp] Processing event.
Meadow StdOut: [esp] Simple event, interface 48, event code: 2099147, status code 0x00000000
Meadow StdOut: [esp] Event data collected, raising event.
Meadow StdOut: [esp] Waiting for event.
Meadow StdOut: [esp] Wifi InvokeEvent NtpUpdateEvent returned CompletedOk
Meadow StdOut: [application] Task ID 1 faulted: One or more errors occurred. (Events not implemented for interface 48)
Meadow StdOut: Inner Exception: Events not implemented for interface 48
Meadow StdOut: [application] Task ID 2 faulted: One or more errors occurred. (Events not implemented for interface 48)
Meadow StdOut: Inner Exception: Events not implemented for interface 48
```

**To Reproduce**
Just run the WiFi basics sample over time, it should take a while to happen. It seems that when the NTP refresh period is lower, it tends to happen more, but might be just a coincidence.

**Expected behavior**
We should not see these garbage events, like this one:
`Simple event, interface 48, event code: 2099147`

**Meadow (please complete the following information as best as you can):**
 - Meadow OS version: 1.12.2.0

bryancostanich, 2024-08-27:
Temp fix is to make this not crash core. May be able to get it into v1.14

ctacke, 2024-09-10:
[PR581 for core](https://github.com/WildernessLabs/Meadow.Core/pull/581) makes this no longer an exception.  It doesn't fix whatever is causing the bad event, but at least now it is just ignored instead of an exception.  Moving priority from P0 to P1 (P2?)

alexischr, 2024-10-29:
I'll try to confirm that the garbage events are gone on long-running applications.

closed 2024-11-26 by alexischr