WildernessLabs/Meadow_Issues #767: CLI Set Meadow time easily

Peter-Moody, 2024-08-30:
**Is your feature request related to a problem? Please describe.**
Currently when setting time via CLI the command looks like this
`meadow device clock 2024-08-19T16:46:24Z`. This is kind of a pain to format and by the time you do format it, it likely has changed.

**Describe the solution you'd like**
It would be good if this could be enhanced to something like
`meadow device clock now` or `meadow device clock utcnow` or `meadow device clock local now`
Which would obtain the time from the PC/Mac CLI is running on and send it to the appropriate connected Meadow.

**Describe alternatives you've considered**
None, this should be pretty easy to implement.

**Additional context**
Only useful for Meadow not connected to a network, like mine usually are.

ctacke, 2024-09-03:
You can actually use

```
meadow device clock now
```

and it will sync with your PC time.  I will update the help doc.

The format of the incoming *should* be any parseable datetime, so I'll see if I can also maybe improve that

ctacke, 2024-09-03:
I just verified you can do something like this and it works:
```
meadow device clock '9/2/2024 11:44'
``

Peter-Moody, 2024-09-03:
Thank you for taking the time to verify. I looked at the docs and `meadow --help` and threw up my hands, I probably could have tried a bit more.

Peter-Moody, 2024-09-03:
Problem was not code but documentation.

closed 2024-09-03 by Peter-Moody