WildernessLabs/Meadow_Issues #778: mscorlib.dll out of sync, etc.

michaelkthompson, 2024-09-24:
Multiple errors in log
...
The out of sync library is: /meadow0/mscorlib.dll
...

Code link sent to support@wildernesslabs.co

Steps to reproduce the behavior:
1. Build and deploy Feather.Sandbox (uses AgChimp.WildernessLabs.Common.Data)
2. Use postman or curl to call Maple locally:

curl --location 'http://<meadow local IP address>:17756/sendcommand' \
--header 'Content-Type: application/json' \
--data '{
    "commandName": "run",
    "args": {
        "iterations": 1,
        "fillTime": 5.0,
        "fillCount": 5000
    }
}'

Expect the code in Feather.Sandbox.SandboxMachineController::Run to run normally

Instead, get these messages in meadow listen:
[application] Run...
[application] DeviceId: 1d-00-36-00-0a-50-30-52-54-31-35-20
[application] Device Name: MeadowF7
[application] IP Address: 192.168.1.227
Inside MachineControllerFactory::Initialize(Sandbox)
[application] POST : /sendcommand --> SendCommand
[application] requestHandlers.Count: 1
[application] Starting up Maple HTTP Request listener.
[application] Maple started on IP Address: 192.168.1.227
[application] Request received from 192.168.1.129:50832
[application] Received POST /sendcommand - Invoking sendcommand
[application] Received command: run
cant resolve internal call to "System.Runtime.Remoting.RemotingS
ervices::IsTransparentProxy" (tested without signature also)
Your mono runtime and class libraries are out of sync.
The out of sync library is: /meadow0/mscorlib.dll
When you update one from git you need to update, compile and ins
tall
the other too.
Do not report this as a bug unless you're sure you have updated
correctly:
you probably have a broken mono install.
If you see other errors or faults after this message they are pr
obably related
and you need to fix your mono install first.
cant resolve internal call to "System.Runtime.Remoting.RemotingS
ervices::IsTransparentProxy" (tested without signature also)
Your mono runtime and class libraries are out of sync.
The out of sync library is: /meadow0/mscorlib.dll
When you update one from git you need to update, compile and ins
tall
the other too.
Do not report this as a bug unless you're sure you have updated
correctly:
you probably have a broken mono install.
If you see other errors or faults after this message they are pr
obably related
and you need to fix your mono install first.
[application] Error running command:  assembly:<unknown assembly> type:<unknown type> member:(null)
[application] Nozzle 1 turned off.

 - OS and version: Windows 11
 - IDE and version: Visual Studio Professional 2022 v17.11.4
 - Meadow extension for IDE version: 1.9.9

Meadow device info:
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 2D-00-2A-00-09-50-30-52-54-31-35-20
    Serial number: 205F318B5230
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 4C:75:25:D6:C9:EC

Firmware Versions
    OS: 1.14.0.0
    Runtime: 1.14.0.0
    Coprocessor: 1.14.0.0
    Protocol: 8

bryancostanich, 2024-09-24:
Tacke has this reproduced, we're on it.

bryancostanich, 2024-10-01:
Have workaround. Still working on this but moving to P-1 since workaround exists.

ctacke, 2025-10-28:
Not seen this is quite some time.  Re-open if it crops back up, but assuming builds between reporting and now have fixed it