WildernessLabs/Meadow_Issues #780: CLIv2 commands e.g. meadow file list /meadow/dir2x/ which as a typo, CLI eventually times out

Peter-Moody, 2024-09-25:
**Describe the bug**
It appears that at least one CLIv2 command, when containing errors, is not reported to CLI by HCOM and the `Concluded` message is not being sent. My guess is that oversights in the recent changes to HCOM, in which subdirectory support was added, is the cause.

**To Reproduce**
1. Use CLIv2 to download a file to subdirectory e.g. `meadow file write -f C:\HcomFiles\Write\Text\TestFile2k.txt -t /meadow0/dir1x/TestFile2k-1.txt`
2. Use CLIv2 to attempt list the file e.g. `meadow file list /meadow/dir2x/` <- Notice a typo `/meadow/` should be `/meadow0/`
3. The above CLIv2 command will eventually timeout. However, if you are viewing syslog output there is a obvious error reported.
    a. `[   64.261000] [ 8] (Error) file/hcom_file_dir_mgmt_utils.c@325-pathname '/meadow/dir2x/' is invalid`
4. It appears that HCOM is not sending this error to CLI and not sending the `Concluded` message either.

**Expected behavior**
CLI should show the error and return immediately.

**Meadow (please complete the following information as best as you can):**
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 34-00-19-00-09-50-30-52-54-31-35-20
    Serial number: 204E31925230
    Coprocessor type: ESP32

Firmware Versions
    OS: 1.14.1.7
    Runtime: 1.14.1.7
    Coprocessor:
    Protocol: 8

adrianstevens, 2025-01-10:
@Peter-Moody let me know if I can assist on the CLI side whenever you want to work on this one

Peter-Moody, 2025-01-10:
@adrianstevens Thanks, I'd totally forgotten about this Issue! I'll let you know if I need help on the CLI side.

closed 2025-01-11 by ctacke

ctacke, 2025-01-12:
Tested as not reproducible on 2.x