WildernessLabs/Meadow_Issues #784: WS2812 cannot color of 1 LED

ctacke, 2024-10-03:
Uncovered while doin ghte ChromaTek button.  Try creating a set of LEDs with any number of LEDs.  the sample of 10 is fine.  Then try setting the color of just 1 of the LEDs.  I definitely fails on LED index 0 or 1, but I suspect fails on any index.

```
leds.SetColor(0, Color.Red);
leds.Show();
```

Often it will do nothing.  Sometimes it will change 2 LEDs, neither to the right color.

This is a hack workaround that might shed light on it:
https://github.com/WildernessLabs/Meadow.Foundation.CompositeDevices/blob/3ea2672f1cda6846ba35d93cdf361cc58ad8ff8f/Source/Switches.ChromaTek/Driver/ChromaTekMomentaryButton.cs#L68

Feels like timing for short comms, since when you address 10 LEDs in one data set, it seems to work fine.

adrianstevens, 2024-10-03:
I can't repro this unfortunately

adrianstevens, 2024-10-16:
shipping v1.1.15.0.0 with a working driver - closing issue

closed 2024-10-16 by adrianstevens