WildernessLabs/Meadow_Issues #791: Allow Bluetooth Server Restart

fredmbarrett, 2024-11-01:
**Is your feature request related to a problem? Please describe.**
Meadow's onboard Bluetooth adapter has a StartBluetoothServer() method that starts the Bluetooth functions of the Meadow platform, however, there is no means by which you can restart the server with a new set of Service/Characteristic definitions. In circumstances where a Meadow-powered device needs to reconfigure the services it announces via Bluetooth (e.g. Sensor configurations changed therefore Characteristics need to change), there is no way short of a device reboot to restart the BLE server.

**Describe the solution you'd like**
BluetoothAdapter.RestartBluetoothServer(serviceDefinition) or BluetoothAdapter.Reset() to clear the way for a new Start() with new Service Definition.

**Describe alternatives you've considered**
Currently, only a Meadow Reboot stops the Bluetooth Server. That is untenable in most production circumstances, as it requires persisting received changes somehow, then rebooting Meadow, and then attempting to restore persisted changes over whatever default circumstances might occur to launch the server at boot time.

**Additional context**
Over-the-air programmable Meadow configuration.

alexischr, 2024-11-26:
This is a duplicate of https://github.com/WildernessLabs/Meadow_Issues/issues/206 - We need a way to stop the bluetooth server, and test that it can be restarted.

NevynUK, 2025-01-20:
ESP code allows restart, support needs to be added to Core.

See issue https://github.com/WildernessLabs/Meadow_Issues/issues/206