WildernessLabs/Meadow_Issues #813: HCOM: cannot list or write files to empty device

ctacke, 2025-01-15:
I have a brand new Meadow, as delivered from the "factory"

`meadow device info` yields:
```
Firmware Versions
    OS: 1.2.0.1
    Runtime: 1.2.0.1
    Coprocessor: 1.2.0.0
    Protocol: 8
```
If I do a `meadow file list` the command hangs indefinitely
If I try to write a file to `/meadow0/update/os` it fails with an HCOM error about a missing directory

Peter-Moody, 2025-01-16:
How to duplicate?

ctacke, 2025-01-16:
I simply opened a new Meadow from the package and tried to do a `meadow file list` - well that was after attempting to write a file.  I wonder if just flashing to the old version would repro as well.

And that gets me thinking that maybe it's trying to use an API that HCOM doesn't support - and perhaps the CLI should just special case and look for early versions and tell the user "update the OS before doing anything else" or something similar?  Not sure what the resolution might be.

adrianstevens, 2026-11-07:
Closing: this was reported against the 1.2 factory firmware. Please update the OS to a current release and open a new issue if listing or writing files still fails.

closed 2026-11-07 by adrianstevens