WildernessLabs/Meadow_Issues #820: Login into Meadow Cloud fails, error -4352

sandervandevelde, 2025-02-03:
**Describe the bug**

I want to connect to the Meadow cloud with the Core model. Device is registered. Settings are set as documented.

This fails.

The logging shows:

```
[esp] Wifi InvokeEvent NtpUpdateEvent returned CompletedOk
[esp] Wifi InvokeEvent NetworkConnectedEvent returned CompletedOk
[esp] WiFi adapter state changed to: Connected
[application] Interface: wlan0: wlan0 Wireless80211 Up
[application] starting health metrics timer
[application] Meadow.Cloud service will use the Esp32WiFiAdapter network interface
[cloud] Attempting to login to https://www.meadowcloud.co/api/devices/login with {"id":"38-00-2F-00-0D-50-4B-55-30-38-31-20"}...
failed to parse private key -4352
[application] Failed to authenticate with Meadow.Cloud
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run the code seen at https://github.com/sandervandevelde/MeadowCloudProject
2. check the logging
3. See error

**Expected behavior**
I expect the connection is made and the logging of the module arrives in the Meadow cloud

**Screenshots**
-

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11
 - IDE and version: Visual Studio Community 2022 v17.12.4
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.9.9.0

**Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7CoreComputeV2
 - Meadow OS version: 2.0.0.3
 - If they are different, please provide these versions as well.
   - Meadow Mono version: ??
   - Meadow coprocessor version: 2.0.0.0

**Additional context**

I tried to connect to the EventGrid Namespace MQTT broker and provided a private key. 

I loaded the latest firmware later on.

Could it be this private key is still available for the code and used here?

Where can I find a list of error codes as this -4352?

How can I remove the private key from the device?

halyssonJr, 2025-02-04:
@sandervandevelde , did you execute those commands?
- meadow login
- meadow provision

alexischr, 2025-02-04:
@sandervandevelde It looks like indeed it is loading a private key from a TLS Client certificate you have previously installed. In order to remove these files, please write empty files to the device to replace them, and reset. Ie. you probably only need to run `meadow file write private_key.pem` where `private_key.pem` is an empty file. Please see https://developer.wildernesslabs.co/Meadow/Meadow.OS/TLS_Client_Authentication/ for more info!

sandervandevelde, 2025-02-04:
I cleared all three files as described in the documentation:

```
meadow file write -f private_key.pem

meadow file write -f client_cert.pem

meadow file write -f private_key_pass.txt
``` 

I stil get the same error:

[image]

alexischr, 2025-04-15:
@sandervandevelde We are still not able to reproduce your problem - I have been doing the following:

```
touch private_key.pem
touch client_cert.pem
meadow file write -f private_key.pem
meadow file write -f client_cert.pem
```

To restore Meadow.Cloud service, and it works on my devices. Attaching a debugger also shows that if empty client certificate files are written on the Meadow, they will be ignored by special logic (which we did have to double-check as it has some edge cases).

We will add some additional debug information to the client cert process that should tell us some more information about what is currently being loaded on your device.