WildernessLabs/Meadow_Issues #823: Anyone notice lost data from GNSS during startup of Clima?

Peter-Moody, 2025-02-05:
**Describe the bug**
It appears a, possibly, modified Clima (version unknown) is dropping bytes from the GNSS module. AFAICT this only happens while Clima is starting up. Since there is no handshaking, this seems possible, especially during Clima startup.

**To Reproduce**
Steps to reproduce the behavior:
1. Start Clima with a logic analyzer connected to the Clima in such a way that the GNSS serial information can be analyzed. This probably means soldering wires on to the correct CCM pins.
2. See error

**Expected behavior**
No errors.

**Additional context**
UARTs have RTS and CTS lines that are intended to prevent overruns, such as this. Since these pins are not considered in the design of our hardware, other means will be used to determine if this problem can be minimized. 

The best candidate is to use DMA. Without DMA the data must be copied from the UART by the MCU before the next byte arrives. With DMA the F7's hardware copies the data from the UART to a buffer.

doingnz, 2025-02-13:
The problem is easily reproduced on either Clima V3.d or V4.x and likely any Meadow F7 with Neo8M GNSS attached by changing NmeaSentance.cs to trace out all GNSS data being received, including the data where the CRC fails. 

The issue does not appear to be related to Clima specifically and not related to other sensor tasks etc. that may be initialising and keeping the CPU busy. There is no requirement to modify any hardware nor to use a logic analyzer. 

The problem can be reproduced with demo app that only initialises the Neo8M and traces the data received.

Steps to reproduce with minimum application are as follows:
1. strip the Sensors.Gnss.NeoM8_Sample down to the minimum code as follows on the develop branch. (must be on develop branch as we need the application to use a modified NmeaSentence.cs.
2. Modify NmeaSentence.cs. to trace out the GNSS data received.
3. Monitor Log for CRC failures and review trace for missing data. 
4. Optionally connect external system to monitor the RX input to the F7 and confirm there is no missing data on the serial IO to he F7. (i.e. confirm the "lost data" is happening in firmware. 

Minimum demo app.
```
using Meadow;
using Meadow.Devices;
using Meadow.Foundation.Sensors.Gnss;
using Meadow.Peripherals.Sensors.Location.Gnss;
using System.Threading.Tasks;

namespace Sensors.Gnss.NeoM8_Sample
{
    public class MeadowApp : App<F7CoreComputeV2>
    {
        NeoM8 gps;

        public override Task Initialize()
        {
            Resolver.Log.Info("Initializing ...");

            //Serial
            gps = new NeoM8(Device, Device.PlatformOS.GetSerialPortName("COM4"), Device.Pins.D05, Device.Pins.A03);

            return Task.CompletedTask;
        }

        public override Task Run()
        {
            gps.StartUpdating();

            return Task.CompletedTask;
        }

    }
}
```

Need to add code to Log output in `Meadow.Contracts\Source\Meadow.Contracts\Peripherals\Sensors\Location\Gnss\NmeaSentence.cs` as follows 
```
            //if (DebugMode) {
            //Console.WriteLine($"sentence        :'{sentence}'");
            //Console.WriteLine($"checksum in NMEA:'{checksumString}'");
            //Console.WriteLine($"parsed checksum :{parsedChecksum:x}");
            //Console.WriteLine($"calced checksum :{calculatedChecksum:x}");
            //Console.WriteLine($"actualData      :{messageData}");
            //Console.WriteLine($"Checksum match? {(calculatedChecksum == parsedChecksum ? "yes" : "no")}");
            //}
            Resolver.Log.Info($"sentence   {(calculatedChecksum == parsedChecksum ? "Ok    " : "not Ok")} : '{sentence.TrimEnd()}'");
            if (calculatedChecksum != parsedChecksum)
            {
                Resolver.Log.Error($"checksum in NMEA:'{checksumString}'");
                Resolver.Log.Error($"parsed checksum :{parsedChecksum:x}");
                Resolver.Log.Error($"calced checksum :{calculatedChecksum:x}");
                Resolver.Log.Error($"actualData      :{messageData}");
                Resolver.Log.Error($"Checksum match? {(calculatedChecksum == parsedChecksum ? "yes" : "no")}");
            }
```
and comment the if to ensure the conditional code is always executed
```
            // make sure data is good
            //if (calculatedChecksum == parsedChecksum)
            {
```
and comment out the else to avoid the exception cluttering the Log.
```
            //else
            //{
            //    throw new ArgumentException("Checksum does not match data. Invalid data.");
            //}
```

Run the demo app and look for the output where the CRC fails. Note the traced `sentance` data and any missing characters.

doingnz, 2025-02-13:
Example output showing two places where data was lost and checksum failed.

```
Using default app.config.yaml...
[core] Log level: Information
[core] MeadowApp
[core] Device is configured to use WiFi for the network interface
[core] All cloud features are disabled.
[application] Initializing ...
[application] sentence   Ok     : '$GNTXT,01,01,02,u-blox AG - www.u-blox.com*4E'
[application] sentence   Ok     : '$GNTXT,01,01,02,HW UBX-M8030 00080000*60'
[application] sentence   Ok     : '$GNTXT,01,01,02,EXT CORE 3.01 (86c0ce)*67'
[application] sentence   Ok     : '$GNTXT,01,01,02,ROM BASE 3.01 (107888)*25'
[application] sentence   Ok     : '$GNTXT,01,01,02,FWVER=UDR 1.50*47'
[application] sentence   Ok     : '$GNTXT,01,01,02,PROTVER=19.20*12'
[application] sentence   Ok     : '$GNTXT,01,01,02,MOD=NEO-M8U-0*7C'
[application] sentence   Ok     : '$GNTXT,01,01,02,FIS=0xEF4015 (200061)*5D'
[application] sentence   Ok     : '$GNTXT,01,01,02,GPS;GLO;GAL;BDS*77'
[application] sentence   Ok     : '$GNTXT,01,01,02,SBAS;IMES;QZSS*49'
[application] sentence   Ok     : '$GNTXT,01,01,02,GNSS OTP=GPS;GLO*37'
[application] sentence   not Ok : 'T,01,01,02,LLC=FFFFFFFF-FFFFFFFF-FFFFFFFF-5FFFFBFE-FFFFFF49*54'
[application] checksum in NMEA:'54
'
[application] parsed checksum :54
[application] calced checksum :5
[application] actualData      :T,01,01,02,LLC=FFFFFFFF-FFFFFFFF-FFFFFFFF-5FFFFBFE-FFFFFF49
[application] Checksum match? no
[application] sentence   Ok     : '$GNTXT,01,01,02,ANTSUPERV=AC *00'
[application] sentence   Ok     : '$GNTXT,01,01,02,ANTSTATUS=DONTKNOW*2D'
[application] sentence   Ok     : '$GNTXT,01,01,02,PF=3BF*4F'
[application] sentence   Ok     : '$GNRMC,,V,,,,,,,,,,N*4D'
[application] sentence   Ok     : '$GNVTG,,,,,,,,,N*2E'
[application] sentence   Ok     : '$GNGGA,,,,,,0,00,99.99,,,,,,*56'
[application] sentence   Ok     : '$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E'
[application] sentence   Ok     : '$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E'
[application] sentence   Ok     : '$GPGSV,1,1,00*79'
[application] sentence   Ok     : '$GLGSV,1,1,00*65'
[application] sentence   Ok     : '$GNGLL,,,,,,V,N*7A'
[application] sentence   Ok     : '$GNTXT,01,01,02,ANTSTATUS=INIT*3B'
[application] sentence   Ok     : '$GNRMC,,V,,,,,,,,,,N*4D'
[application] sentence   Ok     : '$GNVTG,,,,,,,,,N*2E'
[application] sentence   Ok     : '$GNGGA,,,,,,0,00,99.99,,,,,,*56'
[application] sentence   not Ok : '$GNGSA,A,$GNTXT,01,01,02,ANTSTATUS=OK*25'
[application] checksum in NMEA:'25
'
[application] parsed checksum :25
[application] calced checksum :1c
[application] actualData      :$GNGSA,A,$GNTXT,01,01,02,ANTSTATUS=OK
[application] Checksum match? no
[application] Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
Failed to process $GNGSA,A,$GNTXT,01,01,02,ANTSTATUS=OK*56
[application] sentence   Ok     : '$GNRMC,,V,,,,,,,,,,N*4D'
[application] sentence   Ok     : '$GNVTG,,,,,,,,,N*2E'
[application] sentence   Ok     : '$GNGGA,,,,,,0,00,99.99,,,,,,*56'
[application] sentence   Ok     : '$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E'
[application] sentence   Ok     : '$GNGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*2E'
[application] sentence   Ok     : '$GPGSV,1,1,00*79'
[application] sentence   Ok     : '$GLGSV,1,1,00*65'
[application] sentence   Ok     : '$GNGLL,,,,,,V,N*7A'
```

doingnz, 2025-02-13:
in the first log, at a minimum the text `$GNTX` was lost prior to `T,01,01,02,LLC=FFFFFFFF-FFFFFFFF-FFFFFFFF-5FFFFBFE-FFFFFF49*54`

in the second error  there is missing data after `$GNGSA,A,` and prior to `$GNTXT,01,01,02,ANTSTATUS=OK*25`