WildernessLabs/Meadow_Issues #826: ST7789 display not working with FT232H expander

HakanL, 2025-02-13:
**Describe the bug**
On my Windows 11 PC I have a FT232H connected to a ST7789 display. When calling `Show()` on the display (MicroGraphics) my application hangs. I have traced it into the TftSpiBase calling `Exchange` on the SpiBus to write the data and in the Ft232H driver it seems that if there is no data available (displays don't send anything back, MISO/CIPO isn't even connected) then the code for the FT232H will just spin, looking for bytes to receive, see this snippet:
```
    internal int ReadInto(Span<byte> buffer)
    {
        var totalRead = 0;
        uint read = 0;

        while (totalRead < buffer.Length)
        {
            var available = GetAvailableBytes();
            if (available > 0)
            {
                Native.CheckStatus(
                    FT_Read(Handle, in buffer[totalRead], available, ref read));

                totalRead += (int)read;
            }
        }

        return totalRead;
    }
```
In the above code `GetAvailableBytes` is always returning 0, so it will spin forever. I believe the root cause is that the TftSpiBase is calling `Exchange` on the SpiBus, instead of `Write`.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the Ft232h_Sample and add this method:
```
async Task TestSPIDisplay(FtdiExpander expander)
{
    var display = new St7789
        (
            spiBus: expander.CreateSpiBus(),
            chipSelectPin: expander.Pins.D0,
            dcPin: expander.Pins.D1,
            resetPin: null,
            135, 240
        );

    var microGraphics = new MicroGraphics(display)
    {
        CurrentFont = new Font12x16(),
        Rotation = RotationType._270Degrees
    };

    microGraphics.Clear();
    microGraphics.DrawText(0, 0, "Loading Menu");
    microGraphics.Show();

    while (true)
    {
        Debug.WriteLine("Sleeping...");

        await Task.Delay(1000);
    }
}
```
2. Run it
3. If you set a breakpoint you'll see that we never get to the `while (true)` loop.
Note that you can run this without a display connected, but you need a FT232H adapter hooked up.

**Expected behavior**
No hanging or spinning loop, beautiful MicroGraphics dislayed on the display

HakanL, 2025-02-13:
Potentially related to https://github.com/WildernessLabs/Meadow_Issues/issues/727

HakanL, 2025-02-17:
(Note that I had the wrong pin numbers in the small example above, when connected to a real device on the FT232 the SPI is on D0 and D1, so those can't be used for CS and DC). Corrected above.

ctacke, 2025-02-19:
I just checked in some SPI changes.  It successfully does a loopback, and can do the self-test procedure on an MFRC522, which requires a lot of exchanges.  Not saying this fixes this issue, but it's certainly behaving better.
https://github.com/WildernessLabs/Meadow.Foundation/pull/1127

HakanL, 2025-03-13:
@ctacke  I tried the new v2.1.0.1 build, and this works now, but it's about 5x slower than with the previous version of the FH232 driver (I don't believe it's related to the change in the display driver, my guess is an issue with the rewrite of the FH232 driver).

closed 2025-03-13 by HakanL