WildernessLabs/Meadow_Issues #830: Improve WiFi connection status messages

adrianstevens, 2025-02-25:
Testing with MeadowF7_v2 project in the validation sln

```
[esp] SendingCommand returned: WiFiAlreadyStarted
[esp] Wifi InvokeEvent NetworkConnecting returned CompletedOk
[application] Failed to connect: The operation has timed out.
[application] Wi-Fi not connected
```

ESP states we're connected but it's failing at the app level.

adrianstevens, 2025-02-25:
Here's another one 

```
[application] Network Connected. IP: 192.168.1.187
[application] Failed to connect: The operation has timed out.
[application] Wi-Fi not connected
```

NevynUK, 2025-02-26:
Cannot reproduce this, output for the SSL test is:

```
[application] Running WiFiSSLLoopTest`1...
[application] Connecting to valid SSID with valid passcode...
[esp] WiFi adapter state changed to: Connecting
[esp] Sending command to connect
[esp] Event received.
[esp] Processing event.
[esp] Complex event, interface WiFi, event code: NetworkConnecting, status code CompletedOk (0x00000000)
[esp] Getting event data for message ID 0x00000006
[esp] Event data collected, raising event.
[esp] Waiting for event.
[esp] Wifi InvokeEvent NetworkConnecting returned CompletedOk
[esp] Event received.
[esp] Processing event.
[esp] Complex event, interface WiFi, event code: NetworkConnectedEvent, status code CompletedOk (0x00000000)
[esp] Getting event data for message ID 0x00000007
[esp] Event data collected, raising event.
[esp] Waiting for event.
[esp] Wifi InvokeEvent NetworkConnectedEvent returned CompletedOk
[esp] WiFi adapter state changed to: Connected
[esp] SendingCommand returned: CompletedOk
[application] Interface: wlan0: wlan0 Wireless80211 Up
[application] Network Connected. IP: 192.168.1.80
```

One thing that was noticed is that the application is configured to automatically start the next work and then it looks like connect is called: https://github.com/WildernessLabs/Meadow.TestSuite/blob/d8c022e419a5a75c6eae1fbc098070b45f477926/src/Validation/Meadow.Validation.F7/WiFiSSLLoopTest.cs#L35

closed 2025-02-26 by NevynUK

adrianstevens, 2025-03-04:
@NevynUK I assume this is something to do with my network - but we should investigate the feedback to the user and see if there's something we can do to better report the status.