WildernessLabs/Meadow_Issues #833: Broken links on docs

adrianstevens, 2025-03-03:
Here: https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/

Urls are wrong 

for example 
https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Audio/Meadow.Foundation.Audio.md/

should be 
https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Audio/

ctacke, 2025-03-18:
Still an issue.  @lamebrain ?

bryancostanich, 2025-04-08:
@patridge i think this is a docFX thing. can you take a look, please?

patridge, 2025-04-14:
To get an update here, it appears quite a bit of the API docs build process has changed since I last looked at it. My current hypothesis is that there are a couple issues.

First, the new system designed to build API docs only [when the title contains `[build-api]` in a pull request title](https://github.com/WildernessLabs/Documentation/blob/develop/.github/workflows/publish-site-prod.yml#L38) has a flaw. The [Action is triggered only on a "push"](https://github.com/WildernessLabs/Documentation/blob/develop/.github/workflows/publish-site-prod.yml#L5-L6) to the respective branch (e.g., main for publish-site-prod). When an Action triggers for a push, I don't believe it is tied to a pull request anymore. So, the `github.event.pull_request` object being checked will not have any data. As a result, none of the DocFX items are run.

Second, the production might have had a version of the docs published that didn't get fully processed through the Docusaurus routing system. Many of the links are to Markdown files rather than the appropriate destination HTML files. For example, the link to "Meadow.Foundation.Audio" [here](https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/) is linking to `https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Audio/Meadow.Foundation.Audio.md/` instead of `https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Audio/`, where it should link.

After some updates to the process derived from the Action I've assembled into a bash script, I've been able to successfully built a local copy of the API docs. It may be possible to add these resulting DocFX-derived Markdown files to the repo so they are processed by the normal Docusaurus build step before being pushed to the hosting storage.