WildernessLabs/Meadow_Issues #837: CLIv2 isn't displaying error text, instead hangs until it times out

Peter-Moody, 2025-03-25:
**Describe the bug**
If an invalid path is given to via CLIv2 (e.g. `meadow file list /asdf`). The CLIv2 will not display the message sent by HCOM and will instead wait and display a timeout message.

**To Reproduce**
Steps to reproduce the behavior:
1. Using CLIv2 
2. enter an invalid path into meadow file list command.
3. Wait a moment and see timeout error
```
C:\Users\pwmoo>meadow file list /asdf
Getting file list from '/asdf/'...
The operation has timed out.
```
**Expected behavior**
Show error sent by HCOM e.g. `Error during download or delete command processing`

**Screenshots**
This is the behavior when using HcomDiagUI

[image]

adrianstevens, 2025-03-27:
thx @Peter-Moody  - this should be an easy fix

adrianstevens, 2025-04-22:
Fixed in v2.1.0 - closing issue

closed 2025-04-22 by adrianstevens