WildernessLabs/Meadow_Issues #839: Long Console.WriteLine messages lose a character

NevynUK, 2025-03-29:
**Describe the bug**

A long WriteLine message will be split into two (or more parts).  A single character is lost when the message is split.

For example:

```
[cloud] making cloud request to /api/events with payload: {"measurements":{"health.cpu_temp_celsius":37.2197802197802,"health.memory_used":1529264,"health.disk_space_used":8402643,"info.os_version":"2.1.0.1","up_time":"01:45:29.8500000","health.battery_percentage":0,"health.total_memory_arena":30457856,"health.total_memory_free_blocks":83,"health.largest_memory_free_block":9522000
"health.total_allocated_memory":20047408,"health.total_free_memory":10410448,"info.coprocessor_os_version":"2.1.0.0"},"description":"device.health","eventId":10,"timestamp":"2025-03-29T09:39:17.5702300+00:00"}
```

There should be a comma at the end of the first line or at the start of the second line.

**To Reproduce**

Create an application that performs the following:

Console.Writeline("[cloud] making cloud request to /api/events with payload: {"measurements":{"health.cpu_temp_celsius":37.2197802197802,"health.memory_used":1529264,"health.disk_space_used":8402643,"info.os_version":"2.1.0.1","up_time":"01:45:29.8500000","health.battery_percentage":0,"health.total_memory_arena":30457856,"health.total_memory_free_blocks":83,"health.largest_memory_free_block":9522000,"health.total_allocated_memory":20047408,"health.total_free_memory":10410448,"info.coprocessor_os_version":"2.1.0.0"},"description":"device.health","eventId":10,"timestamp":"2025-03-29T09:39:17.5702300+00:00"}");

alexischr, 2025-04-01:
This needs further investigation:

* Was the character lost in Meadow.OS/HCOM?
* Was the character lost in the serial transport?
* Was the character lost in Meadow.CLI?

NevynUK, 2025-08-07:
Updated the Console.WriteLin to:

```
Console.WriteLine("[cloud] making cloud request to /api/events with payload: {'measurements':{'health.cpu_temp_celsius':37.2197802197802,'health.memory_used':1529264,'health.disk_space_used':8402643,'info.os_version':'2.1.0.1','up_time':'01:45:29.8500000','health.battery_percentage':0,'health.total_memory_arena':30457856,'health.total_memory_free_blocks':83,'health.largest_memory_free_block':9522000,'health.total_allocated_memory':20047408,'health.total_free_memory':10410448,'info.coprocessor_os_version':'2.1.0.0'},'description':'device.health','eventId':10,'timestamp':'2025-03-29T09:39:17.5702300+00:00'}");
```

Output:

```
[cloud] making cloud request to /api/events with payload: {'measurements':{'health.cpu_temp_celsius':37.2197802197802,'health.memory_used':1529264,'health.disk_space_used':8402643,'info.os_version':'2.1.0.1','up_time':'01:45:29.8500000','health.battery_percentage':0,'health.total_memory_arena':30457856,'health.total_memory_free_blocks':83,'health.largest_memory_free_block':9522000
'health.total_allocated_memory':20047408,'health.total_free_memory':10410448,'info.coprocessor_os_version':'2.1.0.0'},'description':'device.health','eventId':10,'timestamp':'2025-03-29T09:39:17.5702300+00:00'}
```

There should be a comma after `9522000`

I updated the Console.WriteLine to use single quotes so that the line can be copied verbatim.