WildernessLabs/Meadow_Issues #842: Unable to monitor frequency on PB14 or PB15

ctacke, 2025-04-14:
Using a Project Lab 4 (same pinout as 3.x) I can successfully capture frequency on the `Mikrobus.PWM` and `Mikrobus.RST` pins, but trying the `TX` or `RX` on the same connector just returns zero.  Calls to the driver return a success result.

Test app showing that hooks up 4 lines, 2 that succeed and 2 that fail is attached.
I tested using a signal generator using a 1.1kHz signal.  the first 2 channels output 1100Hz (or thereabout, anyway).  The second 2 channels always return 0.

[FrequencyCounterTest.zip](https://github.com/user-attachments/files/19738722/FrequencyCounterTest.zip)

Peter-Moody, 2025-07-03:
Two things must be true for successfully using `PB14` & `PB15`
1. No ST-Link connected to our debug header.
2. Do not execute the CLI command `meadow uart trace enable`.

alexischr, 2025-07-10:
Fixed with https://github.com/WildernessLabs/Meadow/pull/736