WildernessLabs/Meadow_Issues #843: CLI 2.1.0: `meadow developer` no longer works

NevynUK, 2025-04-18:
**Describe the bug**

`meadow developer --param 0` generates the following output:

```
Setting developer parameter 0 to 0
Available tests:
Available tests:
Available tests:
Available tests:
```

OS version: 2.2.0.0

**Expected behavior**

With no tests built into the system the output should be:

```
Setting developer parameter 0 to 0
Available tests:
    No tests built into the system.
```

adrianstevens, 2025-04-22:
Fixed in v2.1.0.1 - closing issue

NevynUK, 2025-04-28:
`meadow developer` no longer displays four lines of `Available tests:` messages.

However, it does not display the available tests :)

You will need a new OS to test this one, version 2.2.1.0.

### Expected Output

```
Setting developer parameter 0 to 0
Available tests:
    1000 - All ESP32 tests
    1001 - ESP32 Load Test using simple web page
    1002 - ESP32 Load Test using a binary file
```

adrianstevens, 2025-04-28:
Ah interesting ... ok I'll investigate. It may be an issue with how HCOM is returning the data but I'll verify.

NevynUK, 2025-04-28:
> Ah interesting ... ok I'll investigate. It may be an issue with how HCOM is returning the data but I'll verify.

You should be getting the following messages:

```
HCOM_HOST_REQUEST_TEXT_ACCEPTED

HCOM_HOST_REQUEST_TEXT_ERROR - Available tests:
HCOM_HOST_REQUEST_TEXT_ERROR - One for each line of the format "    %d - %s"

HCOM_HOST_REQUEST_TEXT_CONCLUDED
```

adrianstevens, 2025-04-28:
That's exactly what I thought .... should this be coming back as errors?

NevynUK, 2025-04-28:
> That's exactly what I thought .... should this be coming back as errors?

Happy to change, let me know what the appropriate message type should be and I'll update.

adrianstevens, 2025-04-29:
@NevynUK  when there are zero tests, does it still send HCOM_HOST_REQUEST_TEXT_CONCLUDED ?

NevynUK, 2025-04-29:
Yes.

adrianstevens, 2025-04-29:
Ok I think we have this fixed in v2.1.0.3

NevynUK, 2025-04-30:
I think you have.  Tests run:

1 - No tests on the board `meadow developer`:

```
Setting developer parameter 0 to 0
Available tests:
    No tests built into the system.
```

2 - Tests compiled into OS `meadow developer`:

```
Setting developer parameter 0 to 0
Available tests:
    1000 - All ESP32 tests
    1001 - ESP32 Load Test using simple web page
    1002 - ESP32 Load Test using a binary file
```

3 - Run test that does not exist `meadow developer --param 1003`:

```
Setting developer parameter 1003 to 0
Test 1003 cannot be found.  Check that the test has been compiled into the system.
```

4 - Run an existing test `meadow developer --param 1000`:

```
Setting developer parameter 1000 to 0
The operation has timed out.
```

Note the above output is expected as this test takes 5 minutes.

adrianstevens, 2025-04-30:
Ok excellent - closing issue - obviously re-open as needed

closed 2025-04-30 by adrianstevens