WildernessLabs/Meadow_Issues #860: Meadow.CLI is not buildable from source.

mitchdenny, 2025-06-28:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Fork: https://github.com/WildernessLabs/Meadow.CLI
2. Clone: https://github.com/mitchdenny/Meadow.CLI
3. Checkout: develop branch
4. cd repo
5. cd Source
6. dotnet build (fails due to missing projects)
7. cd Meadow.CLI
8. dotnet build (fails due to missing source files (CircularBuffer.cs)

**Expected behavior**
Should be able to build repo from source so fixes can be submitted upstream.

**Screenshots**
N/A

**Developer tools (please complete the following information as best as you can):**
N/A

**Meadow (please complete the following information as best as you can):**
N/A

**Additional context**
Was experiencing some problems with the Meadow CLI so was going to build from source and debug the issue and submit a fix.

ctacke, 2025-06-28:
This is an artifact of how we develop internally.  We tend to always use project refs because often we need to step into other library source that we own.  This project (like nearly all) has project refs to other Wilderness labs repos.  There are 2 paths to get it to compile:
1. remove the project refs and re-add them as NuGet packages.  In this case it's pretty simple - just 3 libraries.  It would be very painful for something like Meadow Foundation which has something like 500 projects and a dozen refs each.
2. pull the source for the other repos.  The CLI itself supports helping with this using the command `meadow source clone` and associated commands.
We definitely need to do a better job of documenting this in each repo, and potentially shipping the internal tool we use to swap refs

ctacke, 2025-07-22:
Readme updated with better instructions

closed 2025-07-22 by ctacke